MarksMakerSpace/filament-profiles #1: :bulb: API endpoint that returns data for requested filament in JSON?

markcarroll, 2024-06-03:
Should it be returned in any particular JSON format? 

Maybe in a format compatible with OrcaSlicer, Bambu Studio or Hue Forge?

davidfischer-ch, 2024-09-11:
That could be a great idea.

wookiee, 2024-09-17:
Yes! Love this idea!

jklewa, 2024-12-24:
I would love to see an official API with documented schemas! This would enable deeper integrations with other projects and streamline the mapping process for popular formats.

For those interested in exploring the datasets further, they can be exported using https://github.com/jklewa/filament-profiles-data

I attempted to wrap the existing filaments schema with a Pydantic data model and started mapping it to Bambu Studio's format: https://github.com/jklewa/filament-profiles-data/pull/1

markcarroll, 2024-12-25:
@jklewa I plan to provide an export to Bambu Studio from filament details or my spools pages.  If you can work out a mapping from my JSON to theirs, I will plug it in.

jklewa, 2024-12-31:
@markcarroll I've figured out most of the mapping (https://github.com/jklewa/filament-profiles-data/pull/1#discussion_r1899901261).

However, before someone prints with them,
1. we should identify which base configs these should inherit from so that they have sane default values
2. someone more familiar with slicer configs should review them

Aptimex, 2025-05-17:
Adding my interest in this feature as well. Would make it possible for your users to change printer filaments by scanning the QR codes they already have on their labels using my [QRSpool project](https://github.com/Aptimex/QRSpool). 

At a minimum the API would need to return the 5 filament data fields in the OpenSpool standard (https://openspool.io/rfid.html#protocol). But to future-proof the API it would also be ideal for it to return all the required fields from the OpenTag standard (https://github.com/Bambu-Research-Group/RFID-Tag-Guide/blob/main/OpenTag.md) since that's probably going to become the de-facto standard once finalized.