MarksMakerSpace/filament-profiles #2: 💡 [FEAT] Add brands feature?

bepstein111, 2024-07-15:
**Is your feature request related to a problem? Please describe.**
I'd like to add a couple filaments I recently analyzed using my new TD-1, but some of them are by popular amazon brands that aren't on your list.

**Describe the solution you'd like**
Ideally, some way for users to add new Brands in addition to new filaments, either as a first step before adding filament, or during the filament adding process.

**Describe alternatives you've considered**
Alternatively, I can give you the brands and they can be added behind the scenes.

**Additional context**
n/a

markcarroll, 2024-07-15:
@bepstein111 Hi, thanks for your input.  I will gladly add that feature. In the meantime, if you give me the names of the missing brands I will add them in right away.

bepstein111, 2024-07-15:
> @bepstein111 Hi, thanks for your input. I will gladly add that feature. In the meantime, if you give me the names of the missing brands I will add them in right away.

Thanks @markcarroll!!
I actually found one of them, Keene Village Plastics was in there but just didn't have anything submitted yet. The other one is 3DGenius.
Thanks again!

markcarroll, 2024-07-15:
ok 3Dgenius is added. 

Thanks for the feedback. I have been working on this for the last couple of months, so it is great to hear people finding it useful.

jbarnhill, 2024-07-22:
any chance kingroon can be added as well??

markcarroll, 2024-07-22:
@jbarnhill Done.

closed 2024-07-25 by markcarroll