MarksMakerSpace/filament-profiles #4: 💡 [FEAT] ability to clone an existing filament entry to add color options

jbarnhill, 2024-07-24:
**Is your feature request related to a problem? Please describe.**
I want to quickly add more "colors" .... and the base options are all the same for this brand/type

**Describe the solution you'd like**
the ability to clone say bambu labs pla basic black and just change the color, would be great

**Describe alternatives you've considered**
so far, it's a lot of copy paste or manually keying the "same info" over and over again

markcarroll, 2024-07-24:
@jbarnhill I will add a clone option, good idea.  Although if you leave all the common parameters empty they will populate from the brand defaults.  If you go to https://3dfilamentprofiles.com/brands/bambu-lab you can set defaults for each filament type for a brand. Those will auto-populate for a new filament color if the same fields are left blank. 

I recognize that the fact I am having to explain this means the UI is clearly lacking.  I will think about some ways to address that.

markcarroll, 2024-10-14:
Expanding on this...

1. Go to the Brands page (top nav bar).
2. Click on the brand you want.
3. Add default settings for brand/material/type if one does not already exist.

Then you can go to the brand/material/type page and add filaments from there.  Doin g it this way you will only need to fill out the color and the ASIN:

So for example for if you created `Kingroom` `PLA` `Basic` you go to https://3dfilamentprofiles.com/filaments/kingroon/pla/basic and click `Add Filament`. Now all you need to do is add the color

[image]

markcarroll, 2024-12-19:
based on your feedback the "My Spools" section has a very front and center "Add more of this color" button. I will get back to the filament input soon.

markcarroll, 2025-04-13:
finally shipped this.

closed 2025-04-13 by markcarroll