MarksMakerSpace/filament-profiles #7: K value limited and printer/extruder/toolhead should be values to enter

mcfool12, 2024-08-12:
I am trying to add some of my filaments to the list and found that the K value only goes out to 3 digits for some reason.  Also the printer/extruder/toolhead should be fields also as values will be different per setup.

mcfool12, 2024-08-12:
Also you are missing Qidi-Tech in the filament list.

mcfool12, 2024-08-14:
Thanks will be getting some filaments added.  Also on the main page where you have the embedded hyperlink to go to the GitHub, you are missing a space between GitHub Repo and or.

markcarroll, 2024-08-14:
Great catch. That is my kind of attention to detail.  Thanks for reporting.  I will fix today.   

One thing.  If you are adding a few colors of the same filament, add the defaults on the brands page first.  Then you can skip that on the filament forms and the defaults will get filled in for you.  Yes I know the UI on that needs improving.

mcfool12, 2024-08-14:
Thanks for the heads up.

davidfischer-ch, 2024-09-11:
I found that K-value depends not only of the hardware, but also of the print speed.

I compared against my own data (eSUN PLA+ mainly) and it look like the "scale" of the value entered in the database is matching the numbers I got from printing Pressure Advance Tower(s) with my high speed profile [1].

Meaning it was tested at ~ 300 mm/s -> Right ?

[1] On a Ender 5 Pro converted to a Mercury One.1

markcarroll, 2024-10-05:
closing this as the original request has been implemented

closed 2024-10-05 by markcarroll