MarksMakerSpace/filament-profiles #9: 💡 [FEAT] Add Checkbox for Bambulab AMS compatibility

Locxion, 2024-09-21:
**Is your feature request related to a problem? Please describe.**
Filter Materials/Colors/Manufactures by AMS and AMS lite compatibility

**Describe the solution you'd like**
Dropdown with
[Blank] - no value
[AMS]
[AMSLite
[AMS&Lite]

**Describe alternatives you've considered**
Some checkboxes maybe?

markcarroll, 2024-10-14:
I've thought about doing this, but the issue I came across was that manufacturers seem to change their rolls between cardboard and plastic without changing the product type. So on one order I got plastic rolls and the next time I got cardboard.  Given that, it seems like adding these would be a moving target.

ksbarnt, 2024-11-22:
This would still be a helpful feature to indicate whether or not the filament *in and of itself* is AMS/Lite compatible. Whether or not it could be dropped in or would need to be adapted/respooled is a separate issue, which I agree would just be a moving target given how often that sort of packaging changes.

markcarroll, 2024-11-23:
Ah, I see what you are getting at.  I will add this to the to-do list for the filament form re-do.

closed 2025-02-25 by markcarroll