MarksMakerSpace/filament-profiles #10: 💡 [FEAT] Add feature to export selected (or all) filaments into CSV file for import into HueForge

jryan15, 2024-10-14:
**Is your feature request related to a problem? Please describe.**
The website is awesome!  No issues.

**Describe the solution you'd like**
Just a feature request to be able to export filaments into a format easy for HueForge to import.  Would be good if the following items could be considered:
 - The ability to only export filaments that have TD values.
 - Ability to check off if the filaments are owned or not (but can do this in HF if needed)
 - If there is a conflict between HF and the website, the website values would likely win due to import function.

**Describe alternatives you've considered**
I exported all the filaments from HueForge and made a spreadsheet to copy/paste profiles into, but it is slow and not easy to update.

**Additional context**
Nope, again awesome job with the site!  Any tips on how I could try and build a site like yours for something different?  I'm a newb to web development, lol.

markcarroll, 2024-10-14:
Thanks for the feedback.  HueForge is something I would like to export to. Do you have a sample CSV for the format I would need to create?

I am also half way through implementing user libraries so you can save the filaments you own.  Should be coming soon, time willing. 

As for the site, if you are just starting out, I would say just start by following one of the many tutorials on Youtube and go from there.  I started this as a way to learn NextJS and fill a need that I saw from the lack of a consolidated site for this info and it grew from there.  Happy to chat about your other site too.  DM me on [Reddit](https://www.reddit.com/r/3dfilamentprofiles/).

jryan15, 2024-10-15:
Awesome Mark, I may take you up on the chat sometime!  I'm attaching a zip file with the HF export formats as well as a couple of images from when I did a test import by just updating a vendor name and couple other things.  I also included a photo of the import tool where he make some specific notes about the import funciton. 

[HF_export.zip](https://github.com/user-attachments/files/17373816/HF_export.zip)
exports from HF.

Thanks again for all you efforts!

ConnorJPSmith, 2024-11-25:
I would also like to see a CSV export option. I am printing labels using NIIMBOT, but right now that process includes downloading the PNG, importing into the label maker app, then printing. If its done via CSV then I could at least select all the filaments I have (based on My Spools) then export a CSV to generate the labels within their app.

bongiozzo, 2025-02-19:
And also exporting Filaments to My Spools from HueForge CSV or JSON

{
    "Filaments": [
        {
            "Brand": "StarPlast",
            "Color": "#000000",
            "Name": "Black",
            "Owned": false,
            "Transmissivity": 0.5000000000000001,
            "Type": "PLA",
            "uuid": "{ed6106f0-4e3a-4e73-8a7f-5531a308ccc3}"
        },
        {
            "Brand": "StarPlast",
            "Color": "#000000",
            "Name": "Black",
            "Owned": false,
            "Transmissivity": 0.5,
            "Type": "PETG",
            "uuid": "{f69f0082-f7f8-4b4e-a293-0cfc623dfd2c}"
        },
    ]
}

markcarroll, 2025-07-14:
Sorry, forgot to update this ticket. I shipped HueForge export at the end of May.