MarksMakerSpace/filament-profiles #12: 🐛 [BUG]

nickglott, 2024-10-22:
I am unable to scan a sigle bar code on my Galaxy S22 or my Galaxy Fold5. They can on my computer but I am thinking they are too small and not enough detail to make it out? Anyone else able to scan them? Maybe just Samsung? have tried the camera app and multiple differet QR scaning apps. 

I know size is really limited but maybe just a little bit bigger or a different QR code style. I like the little spool in the middle haha but maybe dots might work better as they would be a little more seperated and less blending together?

I am using the Phomemo M110 and they look to be detailed but maybe just blending in a bit too much? Maybe I am the only one, not really any settings on the printer can change except darkness have tried 6 and 10.

markcarroll, 2024-10-23:
Hey @nickglott , can you try this:
[image]

markcarroll, 2024-11-01:
@nickglott I deployed a simplified QR code to production yesterday. Let me know how the new one works for you.

nickglott, 2024-11-02:
@markcarroll Sorry been very busy! QR scanning app did read the new QR codes, samsung camera app did not however not the 1st time I have had trouble scanning w/ the camera app. So I think it is good now. Appreciate the effort!

closed 2024-11-02 by nickglott