MarksMakerSpace/filament-profiles #13: 🐛 [BUG]

sreim, 2024-10-28:
**Describe the bug**
Error adding a new entry to the database. It says "There was an error adding the filament. Please try again."
In the console the error message is: "Error adding filament invalid input syntax for type bigint: "NaN""

**To Reproduce**
Steps to reproduce the behavior:
1. Log in using Github account
2. Try to add the following:

1_color: Dark Twinkling Blue
1_brand_id: esun
1_brand_name: eSUN
1_material_id: pla
1_material: PLA
1_material_type_id: ---other---
1_material_type: -- Other --
1_rgb: #053E66
1_fan_speed_min: 0
1_fan_speed_max: 100
1_temp_min: 200
1_temp_max: 240
1_bed_temp_min: 45
1_bed_temp_max: 60
1_softening_temp: 52
1_purchaseLink: https://esun3dstore.com/products/epla-magic?VariantsId=12188
1_ASIN: B0CT2736Z1
0: [{"data":{},"success":false},"$K1"]

**Expected behavior**
Entry to be added. 

**Desktop (please complete the following information):**
 - OS: MacOS
 - Browser Chrome

markcarroll, 2024-10-28:
@sreim thanks for the report.  Something weird is going on as the filament has been added to the database, 5 times now in fact.   I will continue to debug.

markcarroll, 2024-10-29:
@sreim that should be resolved now.

closed 2024-10-29 by markcarroll