MarksMakerSpace/filament-profiles #14: 💡 [FEAT] Multiple/Region links to product

StefanKnol, 2024-10-29:
**Is your feature request related to a problem? Please describe.**
Currently there is only one link you can input, when a shop can have different regions to buy the filament/items depending on where you are, like with Bambu labs, there are 8 regions to buy from. And Amazon has several regions as well.
For example, there were a few leading to a German site. 

**Describe the solution you'd like**
Multiple fields to put in links to the product with a Region field in front.
Standard first field for a global link and then optional extra, max 5.
[REGION] [LINK]
example:
[GLOBAL] [https://store.bambulab.com/products/pla-glow?id=41137288773747]
[EU] [https://eu.store.bambulab.com/en-nl/products/pla-glow?variant=47396823597404]
[DE] [https://eu.store.bambulab.com/de/collections/bambu-lab-3d-printer-filament/products/pla-gluhen?variant=47396823597404]
etc.

**Describe alternatives you've considered**
Clearing up that the link needs to be globally accessible (leading to an English page at least). And not specific for the person putting it in.

markcarroll, 2024-11-01:
great idea.  I have been away of the different country issue and plan to address when I redo the add filament UI. I will incorporate your feedback.

markcarroll, 2025-02-28:
this is coming along. just wanted to give you an update. 

[image]

markcarroll, 2025-03-17:
I have deployed multi-region support for Amazon links.  The product link one will come later.