MarksMakerSpace/filament-profiles #15: 💡 [FEAT] Hi there, great site with good overview. Can you add "Lite" as a seperate material please? eSun PLA-Lite is a bit different to the eSun PLA Basic, so it would be nice to have this material in that list. Thanks a lot in advance.

11Chris11, 2024-11-07:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

11Chris11, 2024-11-07:
Hi there, great site with good overview. Can you add "Lite" as a seperate material please? eSun PLA-Lite is a bit different to the eSun PLA Basic, so it would be nice to have this material in that list. Thanks a lot in advance.

11Chris11, 2024-11-07:
> Hi there, great site with good overview. Can you add "Lite" as a seperate material please? eSun PLA-Lite is a bit different to the eSun PLA Basic, so it would be nice to have this material in that list. Thanks a lot in advance.

Just found the possibility to add the material on my own, ty

markcarroll, 2024-11-07:
Added. LMK if you need anything else.

closed 2024-11-07 by markcarroll