MarksMakerSpace/filament-profiles #16: 🐛 [BUG] Some filaments without logos do not display text in labellife

Arachnid, 2024-11-11:
Filing a bug request as per this thread: https://www.reddit.com/r/3dfilamentprofiles/comments/1gnffo8/manufacturer_logos_not_showing_up_in_labellife/

I've attached filament profiles before and after making the text reappear.
[filament-fiberlogy-pctg-Translucent-Transparent.zip](https://github.com/user-attachments/files/17707811/filament-fiberlogy-pctg-Translucent-Transparent.zip)

markcarroll, 2024-11-15:
@Arachnid It is definitely the fonts. It looks like my Mac's Apple font got written in but you are on Windows.  I think I have picked a universal font now.  Can you give this one a try please. 
[filament-fiberlogy-pctg-Translucent-Transparent new.aml.zip](https://github.com/user-attachments/files/17782324/filament-fiberlogy-pctg-Translucent-Transparent.new.aml.zip)

markcarroll, 2024-11-24:
This should be fixed now. I am specifying a cross-platform font for all the text boxes. Re-open if you are still having an issue.

closed 2024-11-24 by markcarroll

wkearney99, 2024-12-22:
I'm trying to print some Inland PETG Basic and the brand text box is not showing text.  

[image]

If I change the font to Tahoma it immediately shows up.  If I select Arial again, it also remains visible.  

I THINK I FOUND THE PROBLEM.  I downloaded this filament as a native AML file and looked at the XML.  
https://3dfilamentprofiles.com/filaments/2047/details

Arial is misspelled as Ariel in a fontInfos-fontName element at line 75