MarksMakerSpace/filament-profiles #18: 🐛 [BUG] PNG Resolution is now much lower

natefelt, 2024-11-23:
**Describe the bug**
I downloaded a few PNG files for printing today. The generated image resolution is now 337x257 vs 871x665. 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to any filament profile
2. Click on 'click here to download as PNG file'

**Expected behavior**
The low res files are unusable for printing labels.

**Screenshots**
[image]

markcarroll, 2024-11-23:
I updated the way I build those images so that it does not require a massive library any more. I have been targeting label printers so rendered the image to match the size of the 40mmx30mm labels.  Can you tell me how you are using this image so I can determine the right way to solve this?

markcarroll, 2024-11-23:
ok can you try now? That should be fixed.

natefelt, 2024-11-23:
Thanks for looking at this so quickly! The new PNG files are larger resolution, but at a glance, most of pixels are a transparent background and not the actual label.

markcarroll, 2024-11-23:
You are definitely seeing something different than me.  Can you drop one of the PNGs in here so i can compare with what I am seeing. Thanks.

natefelt, 2024-11-23:
Your comment made me realize I should probably test with a different browser. The problem only seems to exist in Firefox (even after clearing cache). It works great in Chromium based browsers.

[image]

markcarroll, 2024-11-23:
hmm, ok thanks for testing that out.  I have tested. in Chromium family and Safari browsers but did not test this feature in Firefox.  I will fire that up and see if I can work out what is going wrong.

natefelt, 2024-11-27:
Tested in Firefox with the recent updates. The issue is resolved. Thank you!

markcarroll, 2024-11-27:
👍 thanks for testing. Hope you like the updates.