MarksMakerSpace/filament-profiles #19: 💡 [FEAT] Add column based filters to the main search screen

ConnorJPSmith, 2024-11-25:
Would like to see a way to filter within specific columns (ie a multi-select for each column) so that you can select a brand/material/etc. Right now, I have been adding some Polymaker PLA+/PRO Polylite filaments I have, but I have to type that over and over as I go back to search to see if they exist. Would be nice to just be able to filter individual columns by existing datapoints

markcarroll, 2024-11-25:
I did play with per-column filters but it took up a lot of screen real-estate.  Maybe I will add it back with a switch to turn off for smaller screens. You can bookmark the search filter as it is now though. 

More importantly, you can go straight to a filtered list like this: https://3dfilamentprofiles.com/filaments/polymaker/pla-plus/polylite

See the attached video. 

https://github.com/user-attachments/assets/feed0e15-d52d-4d51-8dab-029f83fbf254

markcarroll, 2024-12-06:
adding all the columns proved problematic for various reasons.  @ConnorJPSmith  Please take a look at the new filter mechanism I have put in and tell me if this works for you.

ConnorJPSmith, 2024-12-06:
Works for me! Thanks!