MarksMakerSpace/filament-profiles #20: 🐛 [BUG] Sunlu filaments have disappeared

Rouk-s, 2024-11-25:
**Describe the bug**
Seems SUNLU filaments have disappeared from global page

**To Reproduce**
Steps to reproduce the behavior:
I was able to find an URL for a sunlu PLA+ :https://3dfilamentprofiles.com/filaments/9/details or PLA basic Silver :https://3dfilamentprofiles.com/filaments/123/details but it is nowhere to be found in global page :https://3dfilamentprofiles.com/filaments?q=sunlu or https://3dfilamentprofiles.com/filaments?q=PLA%2B

**Expected behavior**
We should be able to find the filament(s) in global page

Tested through Chrome on Android and windows, and Firefox on Windows

markcarroll, 2024-11-25:
Thank you for reporting this.  Turns out the database has a limit of 1000 rows per query and we have now exceeded that. This is now fixed. Sunlu and every thing past that is now back.

closed 2024-11-25 by markcarroll

Rouk-s, 2024-11-25:
thanks @markcarroll