MarksMakerSpace/filament-profiles #22: 💡 [FEAT] In the search, in the 3 dot menu on the right side, add a "Add to My Filaments" quick action.

mharris1984, 2024-11-25:
Like in the title, add a quick action for "Add to my Filaments" in the 3dot menu in the search results.  

Or a checkbox column for selecting multiple results and do bulk add to my filaments.  And if it already exists in the filaments, then add a spool to the spools on hand. 

This way, when I get a bulk order of 15 new filimants to add new colors to my collection or refil low running existing filaments on hand, its a simple one click easy action.  Not having to search, select, click add, back to search, select, click add, etc....

markcarroll, 2024-11-26:
I had the 3dot menu option on the list but not the multi-select yet.  Added.

mharris1984, 2024-11-28:
Awesome!  Thank you!  Building my profile out with my 60 or so spools is a pain adding one by one.  But loving the information all in one spot!

markcarroll, 2024-12-06:
closing this as the menu option is now there. feel free to open a new one if you need more or you find a problem.

closed 2024-12-06 by markcarroll

markcarroll, 2024-12-11:
adding one more comment - @mharris1984 go take a look now. This should be a lot easier, no page navigation required.