MarksMakerSpace/filament-profiles #25: 💡 [FEAT] Table in details for influence of temperature on other parameters

fah, 2024-11-29:
Aero and other lightweight materials that work with foaming require a table that describes the influence of temperature on other parameters.

This can be a lot of work. So I suggest to put this tables on github and work with pull requests. Maybe some manufacturers have API endpoints.

**Solution**
Add tables to the lower part of the detail page.

**Link**
https://eu.store.bambulab.com/collections/bambu-lab-3d-printer-filament/products/asa-aero

Here you can see all the tables I would expect.

markcarroll, 2024-11-30:
This is very interesting, I had no idea about this filament other than it existed.  I will add this to the list for getting on the site.  Are you aware of other manufacturers of this data?

fah, 2024-12-01:
> Are you aware of other manufacturers of this data?

This was one if the best examples. Other Aero or lite materials have the same but not as detailed as this.

markcarroll, 2024-12-02:
I am not all that familiar with Aero brands.  It has taken a lot of hunting with some manufacturers to find their data.   As I said, searching for TDS seems to have given the best results.  

When I get a chance I will have a poke around.

markcarroll, 2025-08-04:
I have added a space to put a link to an external document. I believe this fixes the issue.  If not, please re-open and help me understand what I missed.  Thanks

closed 2025-08-04 by markcarroll