MarksMakerSpace/filament-profiles #26: 💡 [FEAT] Feature to edit Brand

xeon1989, 2024-12-01:
**Is your feature request related to a problem? Please describe.**
I loved the ability to create our own brand in the list.
However, I realized  that I have made mistake in the brand name and not able to correct it. 

**Describe the solution you'd like**
It would be nice to have the feature to edit brand with one of the following way:
- The brand creator able to edit brand created by them, or;
- All users able to suggest (or vote) correction to the name
- Admin to edit or delete brand based on request.

**Describe alternatives you've considered**
No alternative available.

**Additional context**
It would be nice to have the ability to add new Material and Material types too.

markcarroll, 2024-12-02:
Can you tell me the brand name and the correct value and i will amend. 

I am planning on adding the same controls as for filaments - creator can delete, etc, but have been prioritizing other features first.

xeon1989, 2024-12-03:
Hi Mark, it would be helpful to just delete this brand created by me. 

https://3dfilamentprofiles.com/brands/tree-st

Again, thanks a lot for this awesome platform you've made

markcarroll, 2024-12-03:
Done.

xeon1989, 2024-12-03:
Thanks again, looking good.

markcarroll, 2024-12-11:
Edit brand buttons are now available for brands you have created. Hope this suits your needs.

xeon1989, 2024-12-11:
Hi Mark, as test I created a new Brand "Chris Test" and trying to delete it.
It shows "successfully deleted" toast message but the Brand still in the list. 
Tried to edit to "Chris Test Edit" and it works. 
Please see attached network trace in .har format. 

[image]

[3dfilamentprofiles.com.har.zip](https://github.com/user-attachments/files/18092897/3dfilamentprofiles.com.har.zip)

xeon1989, 2024-12-12:
Hi Mark, I tried this morning (Asia time) and the deletion now works. 
As always, thanks a lot for delivering the feature in such short time.
Cheers. 

Chris