MarksMakerSpace/filament-profiles #27: 🐛 [BUG] Filtering not working in "Brand / Defaults" and "Materials"

BlackRose01, 2024-12-04:
**Describe the bug**
The function to filter the list will always show "No items found".

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Brand / Defaults' or 'Materials'
2. Click in Search Field
3. Type any filterable value in the field (e.g. SUNLU or sunlu) - Capitalization is uninteresting
4. Result: "No items found"

**Expected behavior**
A list with the filtered result

**Screenshots**
[image]
[image]

----

[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser firefox
 - Version 133.

markcarroll, 2024-12-04:
crap, I think I broke this last night. thanks for catching @BlackRose01

markcarroll, 2024-12-06:
this should now be fixed.

BlackRose01, 2024-12-06:
@markcarroll Sorry for reopening the issue but the problem still exists on the Materials page
[https://3dfilamentprofiles.com/materials](https://3dfilamentprofiles.com/materials)

markcarroll, 2024-12-07:
hey don't apologize. I appreciate the assist.  I am signing you up as my official test team. 

That page is fixed and deployed now. Thanks again.

closed 2024-12-07 by markcarroll