MarksMakerSpace/filament-profiles #30: 💡 [FEAT] Possibility to add Manufacturer Number and EAN/GTIN

BlackRose01, 2024-12-04:
**Describe the solution you'd like**
It would be nice to have some additional field for the filament. The fields should be optional when you create a filament. If the field is emtpy on creation any user can update the field but when there is a value in then the field is readonly.

The following fields would be very nice to have:
1. Manufacturer Number: Unique Number from the manufacturer - Normally changed from manufacturer on upgrade
2. EAN (European Article Number) - Unique Number worldwide for traded products, and others (replaced by GTIN)
3. GTIN (Global Trade Identification Number) - Unique Number worldwide for traded products and others (stronger, better, faster)

GTIN and EAN can be combined in one field.

**Additional context**
GTIN and EAN are optional IDs provided from different organizations but is highly recommend (mandatory for Amazon, eBay and others). With such an ID it is possible to search for a product in offical databases.

BlackRose01, 2024-12-05:
Please also add the a field for Tolerance

markcarroll, 2024-12-06:
Added to the list

markcarroll, 2025-02-18:
I released this today. It is just one combined field for now, we can extend if we need to.

closed 2025-02-18 by markcarroll