MarksMakerSpace/filament-profiles #36: 💡 [FEAT] Multi-user Support in "My Spools"

luckygreen, 2024-12-18:
**Is your feature request related to a problem? Please describe.**

At our 3D printing enthusiasts club, we would like to know which spools we have in inventory by storing them in "My Spools". Spools that should be shown in "My Spools" come in two categories:

- Spools owned by the club that anyone may use at will on our club-owned X1C. Any club member should be able to check in, check out, and update quantities on hand in "My Spools" by using their own GitHub account, rather than some shared GitHub account. (Since members may join and leave).

- Spools that are owned by an individual member, but that the owner in general and within reason is willing to provide use of to other members. Those spools should also be listed in the club's "My Spools", with a field as to which member (=GitHub account) one has to ask for permission to use some filament off an individually owned spool.

We are operating in a collaborative environment. There is no need for a complex permissions system.  When checking in a spool, a member of the project (you can think of the members as committers) should be able to tag a spool as either club owned or owned by another GitHub account. If the solution requires for the club to have a GitHub account that can be tagged as the owner, that's fine.

**Describe the solution you'd like**
- Enable multi-user access to "My Spools"
- Add a field to "My Spools" listing the owner of each spool
- 
**Describe alternatives you've considered**
We considered having a club "My Spools" account for club-owned spools, with each member listing their spools willing to be shared via the existing "My Spools". That would require members to maintain two "My Spools" listings: one for spools that they aren't willing to share and one for spools they are willing to share.

That approach is not workable, for two reasons:
- Each individual owner would need to maintain a list of all their spools and  a second list of spools they are willing to share under two different GitHub accounts.
- A club member looking for a particular type of filament would have to manually search the club's "My Spools" as well as the "My Spools" of each member making spools available. There is no unified view. Until "My Spools" has a concept of multiple owners within one "My Spools" list.

**Additional context**
Please do not hesitate to ask me for any clarifications or to validate potential solutions.
Thank you for your consideration!

luckygreen, 2024-12-21:
For a long list of common use cases that the absence of this feature precludes, see the comments in what amounts to the analogous feature request for Bambu Studio:
[BambuStudio connect multiple accounts to one printer #3533](https://github.com/bambulab/BambuStudio/issues/3533).

markcarroll, 2025-02-28:
This is not everything you asked for, yet, but here is a start:
https://www.reddit.com/r/3dfilamentprofiles/comments/1j01i8w/sharing_of_spools_and_colors/ 

[image]