MarksMakerSpace/filament-profiles #37: 💡 [FEAT] Add Filament Dryer: Graef DA-2042 Dörrautomat

luckygreen, 2024-12-18:
**Is your feature request related to a problem? Please describe.**
As a San Francisco expat living in Berlin, I quickly discovered that the **Graef DA-2042 Dörrautomat** is _widely_ used in Germany as a filament dryer. The reason being not just its low price, but the fact that this high-airflow fruit dryer goes up to 80C, which is required to release the water bonds in Nylon. While most commercial filament dryers only go up to 75C, which is insufficient to dry Nylon. No matter how many hours you keep the Nylon at 75C.

**Describe the solution you'd like**
Add the _Graef DA-2042 Dörrautomat_ as a filament dryer, since it is widely used as such.
https://www.amazon.de/Graef-DA-2042-D%C3%B6rrautomat-schwarz/dp/B08YR1MQ1F

**Describe alternatives you've considered**
The alternative is that you do not list what is one of the most popular filament dryers in this part of the world.

**Additional context**
While I haven't looked at every filament dryer on the market, I have never seen one that goes above 75C. Which, unbeknown to most buyers, makes the Graef DA-2042 one of the few filament dryers actually capable of drying all filaments.

markcarroll, 2025-01-07:
I would love to add this one but I will have to wait until I finish my update to support checking non-US amazon sites for prices.  I am working on it though, 21% of my users come from Germany, so it is important to me to support them.

markcarroll, 2025-07-14:
I am not listing food dehydrators at this time. I may add later.