MarksMakerSpace/filament-profiles #38: 💡 [FEAT] Add Shrinkage Field

ShockinglyMoist, 2024-12-18:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
I would like to see an optional 'shrink %" field for those of us who make detailed functional prints

**Describe alternatives you've considered**
Currently, after I do my calibration run on a filament, I make a label and stick it on the reel. 

**Additional context**
An alternate solution could be adding a "Tips/Notes" text field to add misc information like this. Users could additionally write a small blurb stating hints or other oddities. Both would be awesome.

markcarroll, 2024-12-19:
I added a notes field to the My Spools section. I will look at adding one to the main filament as well.

markcarroll, 2025-05-13:
Closing this now that I have added a notes field and per-spool QR code labels.  Hopefully that works for you.

closed 2025-05-13 by markcarroll