MarksMakerSpace/filament-profiles #39: 🐛 [BUG] Unable to remove a filament from My Spools

jklewa, 2024-12-18:
**Describe the bug**
I cannot remove a spool from My Spools once it has been added. There are also issues with typing numbers into the 'Number of spools' input field.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to a filament details page 
2. In 'Number of spools of this type/color:', click on '-' until `0`
3. Refresh the page
4. 'Number of spools' is still `1`

**Expected behavior**
* Removing a spool (setting its count to `0`) should remove it from the 'My Spools' list
* Typing in the number of spools should be saved
* While typing, I should be able to delete the previous number so that I can type a new one

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser chrome
 - Version 131.0.6778.140

markcarroll, 2024-12-19:
good timing. I just deployed a big update. take a look and let me know what you think.

jklewa, 2024-12-19:
Oh, wow! I see the 'Beta' flag is gone, and the new spool management allows me to delete spools!

closed 2024-12-19 by jklewa

markcarroll, 2024-12-19:
the paint is still dripping wet, as they say. so please let me know if you find any issues.