MarksMakerSpace/filament-profiles #40: [SUGGESTION] Spool Identifier

BlackRose01, 2024-12-19:
Hey,

maybe this could be an awesome integration in the website. The mentioned problem (closed RFID Tag System by Bambu Lab) is real but I think it is also very cool for other printer systems. Maybe there are 2 use cases here:

1. Nice link to the website that you can build the system on you own
2. The possibility to generate the necessary information for the RFID Tag on the website

I think it is also nice case for a cooperation.
As an example: From your side there could be a connector for the OpenSpool Hardware to fetch a JSON list of available spools (filterable - manufacturer, type) and then get the spool information in the required format for the RDIF Tag. The OpenSpool Guys could enhance there OSH Tool with a display to scroll through/filter your list and then select a spool for writing the tag.

Maybe this side project requires then the implementation of a API token:
1. Security Purposes
2. Monetization (Web hosting, Database incl. Domain can be expensive over time)
3. Access user areas (Maybe i don't need the whole list - Usage of My Spools) - Saves also ressources
4. Enhancement for later implementations or use cases (maybe connectors for companies - Point 2)

KR,
BlackRose

---

[1] [https://openspool.io/](https://openspool.io/)
[2] [https://github.com/spuder/OpenSpool](https://github.com/spuder/OpenSpool)

BlackRose01, 2024-12-19:
I saw that there is already an issue opened by somebody else. Here is the issue:

https://github.com/spuder/OpenSpool/issues/13

markcarroll, 2024-12-20:
thanks for the suggestion.  I will investigate when I get a chance.

markcarroll, 2025-03-17:
we now have a link for SKU/Other id.  I have successfully searched for filaments using a bar code scanner so hopefully this will help.