MarksMakerSpace/filament-profiles #41: 🐛 [BUG] Table Flickering on changing Table Page Limit

BlackRose01, 2024-12-19:
**Describe the bug**
The bug can be triggered in two different ways. When it happens then the table begans to flickering. To stop that you need to reload the page or change to another web page.

**To Reproduce - First Way**
Steps to reproduce the behavior:
1. Go to 'Filament List'
2. Select a Table Page Limit (wait until the limit is visible as query parameter in url)
3. Click on 'Filament List' again
4. See error

**To Reproduce - Second Way (hard to trigger)**
Steps to reproduce the behavior:
1. Go to 'Filament List'
2. Select a Table Page Limit (wait until the limit is visible as query parameter in url)
3. To to next page or change the Limit again
4. See error

**Additional Information**
This bug is hard to trigger. It seems that the Page is in an infinite loop to switch between two pages or the page tries to reload the page without the limit information, then it adds the limit to the url and that forces a reload of the site. 

**Expected behavior**
There should be no flickering.

**Desktop (please complete the following information):**
 - OS: Windows 11
 - Browser firefox
 - Version 133.0.3 (latest version)

**Videos**
[1]
https://github.com/user-attachments/assets/8e981d48-2c23-42b7-a0f2-dfa26ce03104

[2]
https://github.com/user-attachments/assets/c1966f04-0ef1-4193-8f72-9a58ea362708

BlackRose01, 2024-12-19:
I tried my best to describe the bug in a good way but it is very hard.

markcarroll, 2024-12-20:
wow that was a bad one. thank you for catching it _and_ logging it for me.  That took a while to solve but I think it is good now. Let me know if you find anything else.

closed 2024-12-20 by markcarroll