MarksMakerSpace/filament-profiles #43: 💡 [FEAT] Filament wish/shopping list

ianwinter, 2024-12-20:
**Is your feature request related to a problem? Please describe.**
No

**Describe the solution you'd like**
Would be good to maintain a wish list or shopping list so you could track filaments you might want to buy but don't yet have. In theory could copy functionality of the "my spools" but just rename.

**Describe alternatives you've considered**
Maintain it in a list locally.

markcarroll, 2024-12-20:
Great idea.  Will look into it

hansenc0705, 2025-09-28:
Just
To add to this, consider an easy reorder type function as well. It's cheaper to buy in bulk say from Bambu so keeping a list of spools needed would be great then once a threshold is hit can easily see what I need to go order. 

Thanks for the work being done on this project!