MarksMakerSpace/filament-profiles #44: 💡 [FEAT] Add EU Bambu Labs Store as source for pricing information

luckygreen, 2024-12-20:
**Is your feature request related to a problem? Please describe.**
Currently, "My Spools" pulls pricing from the U.S. Bambu Labs store only. This doesn't do me much good, since I am an expat living in Germany. 

**Describe the solution you'd like**
Add support for https://eu.store.bambulab.com/ as a source of pricing information.

**Describe alternatives you've considered**
Clicking onto the USD price in the GUI actually does redirect me to the EU Bambu Labs store. Presumably due to geolocation. I can look up the current price  in EUR by clicking on the price in USD.

**Additional context**
Feel free to contact me with any questions or if you need a VPN exit in Germany to test this feature. I will be happy to provide you with one.

markcarroll, 2024-12-20:
yes, international support is on my list. Over 20% of my users are in Germany so it has not gone unnoticed. I did make sure the links get redirected to the regional store as you have noticed, but the I get priced they are pulled from the US one. I will let you know when I get to it.

luckygreen, 2024-12-21:
Good job having put the redirect in place!

To not give the wrong impression, this feature would be a convenience. Its absence doesn't significantly impact broad swaths of common use cases with no workarounds. Unlike #36, which is 100x as important as this feature request here.

markcarroll, 2025-03-17:
I have added country support for the Amazon pricing system.  I will continue rolling through adding country support to the other sections too.