MarksMakerSpace/filament-profiles #46: 💡 [FEAT]  Please add JPEG for newly-added filament

luckygreen, 2024-12-21:
**Is your feature request related to a problem? Please describe.**
A user can't manually upload an image to be associated with a newly-added filament. (For understandable reasons). Which means the admin has to do it by hand. ;-)

**Describe the solution you'd like**
Display the attached JPEG for the following filament that I just added, which is no longer being manufactured:
https://3dfilamentprofiles.com/my/filaments/details/1940

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

markcarroll, 2024-12-22:
done

closed 2024-12-22 by markcarroll

luckygreen, 2024-12-22:
👍️
Now the page looks "complete".