MarksMakerSpace/filament-profiles #48: 💡 [FEAT] LARGE FORMAT TEXT

wkearney99, 2024-12-22:
**Is your feature request related to a problem? Please describe.**
It's difficult to see the small white text on the black background.

**Describe the solution you'd like**
I want to use the labels to let me quickly pick spools off the shelf.  If the text for the material type was larger it would be a LOT of help.

**Describe alternatives you've considered**
Re-arrange the label to better emphasize the material type.

**Additional context**

markcarroll, 2024-12-22:
The text resizes to fit the amount of letters needed for the material so for longer materials it will get smaller.  One option here is to change the download type to "Native". That will give you a label with each separate field that you can change in Labelife.  If you find a layout that works well send it over and I will see how it fairs when I run all the different types through it.

wkearney99, 2024-12-22:
I posted about it, and some pix on the subreddit.   Don't know if it's better to have the conversation here, or there: https://www.reddit.com/r/3dfilamentprofiles/comments/1hjofma/make_the_labels_easier_to_read/

I understand how the text size gets reduced as the string gets longer. _Once upon a time I wrote software that had to deal with internationalization... don't get me started on how looooong German words are in dialog boxes..._ 

If the box ran full-width that'd open up some room for larger text.  I just want to be able to see it better from a distance.

markcarroll, 2025-08-14:
@wkearney99 I made the text a lot bolder now. see if that helps. 

BTW, I used to work on Windows 2000 & XP so I remember the German internationalization problem.