MarksMakerSpace/filament-profiles #54: 🐛 [BUG] MySpools Page not showing pricing.

mikebronner, 2024-12-22:
**Describe the bug**
I would love to see pricing on My Spools page, that would quickly let me see if the current price is reasonable. What would be even better would be to indicate if the price is lower than average for that specific filament, indicating it might be on sale and I could quickly stock up while the price is low.

Thanks!

jklewa, 2024-12-23:
I believe this is caused by a data serialization issue. Inspecting the RSC requests, the price_data is a string that contains the actual json object.

There are also "[object]" strings in some unexpected places. Those also show up on the main filaments page - so not strictly related to this issue.

markcarroll, 2024-12-23:
Thanks for the report. I dug in and realized that I had to convert a json column to text to build the summary table. converted it back client side so the price data is properly extracted now.  

@jklewa I've seen some of those [object] strings around too but can't work out how they got there. still trying to track that one down.  Closing this as the main issue is now solved.

closed 2024-12-23 by markcarroll

mikebronner, 2024-12-23:
awesome, thank you for fixing this so quickly, I now see prices as expected. :)