MarksMakerSpace/filament-profiles #55: [Suggestion] Manufacturer addition request

k-piekarczyk, 2024-12-22:
If possible, please add Spectrum Filaments to the list of manufacturers.

Link to their webpage: https://spectrumfilaments.com/en/

markcarroll, 2024-12-22:
@k-piekarczyk  `Spectrum` is already there. Should we change the name to `Spectrum Filaments`?

k-piekarczyk, 2024-12-22:
Ah, no need to do that, I just didn't notice it's there 😅 Thank you for a quick response, I'll close the issue now.

k-piekarczyk, 2024-12-22:
@markcarroll though while I have you here, it would be awesome if you added the `Pastello` material type (or the ability to add own material types, but it's a bigger ask), here's an example: https://shop.spectrumfilaments.com/product-eng-1514-Filament-Pastello-PLA-1-75mm-FLAMINGO-RED-1kg.html

markcarroll, 2024-12-23:
done

closed 2024-12-23 by markcarroll