MarksMakerSpace/filament-profiles #59: 🐛 [BUG] Missing Fillerment Type

LostkingZ, 2024-12-23:
**Describe the bug**
Please add eSUN PLA-ST as it is missing.

https://www.amazon.co.uk/eSUN-Filament-Dimensional-Accuracy-Printing/dp/B08MW1LXQY/ref=sr_1_1_sspa?crid=1UN1EGDYH8RV3&dib=eyJ2IjoiMSJ9.Q-oVS5CS3FfV9g5xyIOaQLVsLukbDhxmjfme6prgAMcRka2Vqp3Z5RikfkeaRwiS8C0bYwki0y5zqSYXLYBn0SC2ZnZ79KUcx2XJnXEAOB-SUfH782bNQmimtSpaMBuoe4b8GNMg_4oSzWpoiDo-tvsg0CjW4VxlRa9qNRj_nqy6PInQdN1_jr14rT4SvGuX833Akag3HniykYpI0gEyJ0juR702LxgIDnt3H7Kq5nIubOzdSxwS-ZqAlqpb478LTgtW97cVbCXwQ8fKDrC_lPie7YMDJdcSYIyZR9-Mncc.9gaYWn2p6tO17suh6VyrDy6uBu3bWWurG7nkuJKuO8Y&dib_tag=se&keywords=PLA-ST&nsdOptOutParam=true&qid=1734967709&sprefix=pla-st%2Caps%2C168&sr=8-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&psc=1

https://www.esun3d.com/epla-st-product/

Thanks. :)

ch33tah0, 2024-12-23:
I'll just hijack this... I missed "APLA" :)
But maybe you could check the list below and add the missing ones to the Material / Material Type list?

eSun: | SUNLU:
-- | --
**High-Speed   Filaments:** | PLA
PLA-HS | PLA+
PLA+HS | ABS
ABS+HS | PETG
TPU-HS | PLA Meta
PETG+HS | PLA Carbon Fiber
  | APLA (Anti-String)
**General   Filaments:** | High-Speed PLA
PLA-Basic | Easy ABS
PLA+HS | TPU
PLA+CMYK | TPU Silk
ABS | ASA
ABS+ | PA6-CF
PETG | Easy PA
PETG-LITE | PC
PET | PCL
TPU-95A | PVB
TPU-64D | High Speed Matte PETG
  | PETG-CF
**Aesthetic   Filaments:** | Silk PLA+
PLA-Rock | Silk PLA+ Rainbow
PLA-Chameleon | Silk PLA+ Dual Color
PLA-Magic | Silk PLA+ Tri-Color
PLA-Silk Magic | PLA Rainbow
PLA-Silk   Mystic | PLA Matte
PLA-Silk | PLA Wood
PLA-Silk   Rainbow | High Speed PLA Marble
PLA-Matte | PLA Color Change   (Temperature)
PLA-Matte Dual | PLA Glow In The Dark
PLA-Matte   Rainbow | PLA Twinkling
PLA-Luminous   Rainbow | PETG Glow In The Dark
PLA-Marble | PETG Silk
PLA-Twinkling |  
PLA-UV Color   Change |  
  |  
**Engineering   Filaments:** |  
ASA+ |  
ABS-HAT |  
ABS-CF |  
ABS-GF |  
PETG-CF |  
PET-CF |  
PA-CF |  
PA12-CF |  
PLA-GF |  
PLA-CF |  
PLA-ST |  
PC |  
PLA-LW |  
ASA-LW |  
TPU-LW |  
PEEK-Industrial |  
PETG-ESD |  
PET-FR |  

Sorted: A->Z
--
ABS
ABS+
ABS+HS
ABS-CF
ABS-GF
ABS-HAT
APLA (Anti-String)
ASA
ASA+
ASA-LW
Easy ABS
Easy PA
High Speed Matte PETG
High Speed PLA Marble
High-Speed PLA
PA12-CF
PA6-CF
PA-CF
PC
PCL
PEEK-Industrial
PET
PET-CF
PET-FR
PETG
PETG Glow In The Dark
PETG Silk
PETG+HS
PETG-CF
PETG-ESD
PETG-LITE
PLA
PLA Carbon Fiber
PLA Color Change (Temperature)
PLA Glow In The Dark
PLA Matte
PLA Meta
PLA Rainbow
PLA Twinkling
PLA Wood
PLA+
PLA+CMYK
PLA+HS
PLA-Basic
PLA-CF
PLA-Chameleon
PLA-GF
PLA-HS
PLA-Luminous Rainbow
PLA-LW
PLA-Magic
PLA-Marble
PLA-Matte
PLA-Matte Dual
PLA-Matte Rainbow
PLA-Rock
PLA-Silk
PLA-Silk Magic
PLA-Silk Mystic
PLA-Silk Rainbow
PLA-ST
PLA-Twinkling
PLA-UV Color Change
PVB
Silk PLA+
Silk PLA+ Dual Color
Silk PLA+ Rainbow
Silk PLA+ Tri-Color
TPU
TPU Silk
TPU-64D
TPU-95A
TPU-HS
TPU-LW

markcarroll, 2024-12-23:
Yeah the simple list i started with seems to have only scratched the surface.  I will work through these today. Thanks for the very comprehensive list. That makes my life a lot easier.

markcarroll, 2024-12-24:
ok, I'm pretty sure I have all combinations of all of those now including @LostkingZ your original request - https://3dfilamentprofiles.com/filaments/esun/pla/super-tough

closed 2024-12-24 by markcarroll