MarksMakerSpace/filament-profiles #60: 💡 [FEAT] Add Filament entries count to "Brands" page

ch33tah0, 2024-12-23:
Add a new column "filament entries" to https://3dfilamentprofiles.com/brands
This may help to identify "good/big" brands or what brand got the most collected filament information?

Something like that?

Brand | #
-- | --
Bambu Labs | 198
Polymaker | 165
SUNLU | 141
eSUN | 106
Elegoo | 82
Overture | 80
Inland | 70
Geeetech | 61
Eryone | 53
Jayo | 53
COEX | 50
Creality | 42
Amolen | 36
Kingroon | 35
Hatchbox | 34
Ziro | 31
FlashForge | 27
Anycubic | 25
3DHoJor | 23
Giantarm | 23
Mika3D | 23
Protopasta | 22
Siddament | 20
Amazon Basics | 19
CC3D | 19
IIID Max | 16
CoLiDo | 15
Ujoybio3d | 14
3DJake | 12
Extrudr | 12
Prusament | 12
iSanmate | 11
Matterhackers | 11
Zyltech | 11
3DKordo | 10
Cookiecad | 10
DURAMIC 3D | 10
R3D | 10
Stronghero3d | 10
OVV3D | 9
Printed Solid | 9
Yousu | 8
Polar Filament | 7
SPOOL3D | 7
3DE | 6
GTMax | 6
MakerBot | 6
Rosa3D | 6
add:north | 5
Botfeeder | 5
CCTREE | 5
DOWELL 3D | 5
Filabotics | 5
HUTI | 5
Likesilk | 5
Numakers | 5
Paramount 3D | 5
Spectrum | 5
VOXELPLA | 5
123-3D | 4
3Dgenius | 4
Ankermake | 4
Bavaria | 4
ColorFabb | 4
Forshape | 4
JAMG HE | 4
PRILINE 3D | 4
Recycling Fabrik | 4
RepRapper | 4
Tinmorry | 4
3D-Fuel | 3
Arianeplast | 3
BING3D | 3
Clas Ohlson | 3
Devil Design | 3
Fiberlogy | 3
Filaments.CA | 3
Fillamentum | 3
GreenGate3D | 3
JUSTMAKER | 3
kexcelled | 3
Polaroid | 3
SainSmart | 3
SmartMaterials | 3
Snapmaker | 3
TEQStone | 3
AzureFilm | 2
Das Filament | 2
DO3D | 2
FilRight | 2
FormFutura | 2
GST 3D | 2
H3D | 2
HP 3DF | 2
IEMAI | 2
iSANGHU | 2
Jarees | 2
Keene Village Plastics | 2
NOVAMAKER | 2
PEAKACE | 2
Solutech | 2
Wanhao | 2
3D Printing Canada | 1
3D Warhorse | 1
Aceaddity | 1
American Filament | 1
Amoybaby | 1
AURAPOL | 1
BASF | 1
Bblife | 1
CMACAIV | 1
Color Me 3D | 1
COLORFUL | 1
Copymaster3D | 1
D3D Sigma | 1
EIBOS | 1
eono | 1
FilaCube | 1
Filamentive | 1
FIROS | 1
FUSED MATERIALS | 1
Gizmo Dorks | 1
GratKit | 1
KINLUOT | 1
Made for Prusa | 1
Maertz | 1
Material4Print | 1
Menetee | 1
MKOEM | 1
Ooznest | 1
PINRUI | 1
PureFil | 1
QIDI Tech | 1
Recreus | 1
RecyclingFabrik | 1
SA Filament | 1
TM3D | 1
TTYT3D | 1
Verbatim | 1
Visdream | 1
XYZPrinting | 1
Yoyi Yoyi | 1

markcarroll, 2024-12-24:
This is done and will show up with the next deployment.  Just out of interest @ch33tah0 where did you get those numbers from?

ch33tah0, 2024-12-24:
I just pulled them into Excel from https://3dfilamentprofiles.com/filaments?perPage=999999 and did some Excel magic to that list. 😂

btw, are the following the only PNG logos you have at the moment (for the label)?

https://3dfilamentprofiles.com/logos/bambu-lab.png
https://3dfilamentprofiles.com/logos/creality.png
https://3dfilamentprofiles.com/logos/elegoo.png
https://3dfilamentprofiles.com/logos/esun.png
https://3dfilamentprofiles.com/logos/overture.png

I'm currently working on a [50*30 label for Niimbot](https://makerworld.com/en/models/49600#profileId-51451) and collecting png logos for my brands.

If you're interested, I could upload them once they are ready. (<100kb, width <1000px, preferably transparent png?)

[image]

Maybe you would also like to have a look at this label file and add it to the download? 😁
[Niimbot_50x30_Bambulab_text_QR.jcps.json](https://github.com/user-attachments/files/18238853/Niimbot_50x30_Bambulab_text_QR.jcps.json) (actually ends with .jcps)
There is only the Bambu logo and some thumbnails in there, the rest is text.

markcarroll, 2024-12-24:
Yes, right now those are all i have. Gathering logos is fairly low on my priority list right now. I do have a feature to upload logo images mostly coded up, but since a bad logo could really mess things up, I haven’t had the time to harden it yet and deploy it.  I will get back to it at some point.  If you have logo files in transparent PNGs though, I will take them.  Just open a ticket at attach the PNGs. 

As for the label, I would love to support more label types and sizes.  I will take a look at what you have sent me, although it may be a while before I get to it.

markcarroll, 2024-12-30:
closing the original ticket. open a new one if you have images to add

closed 2024-12-30 by markcarroll