MarksMakerSpace/filament-profiles #61: 🐛 [BUG] Spools added to "My Spools" sometimes do not get added to "My Spool"

luckygreen, 2024-12-23:
**Describe the bug**
I've noticed since using the site that sometimes selecting a filament and clicking on "My Spool" does not add the spool to "My Spool". I just had it happen 4x in a row and still don't have the spool added.

**To Reproduce**
Steps to reproduce the behavior:
1. Select a filament from the Filament List. I tonight's instance Bambu 70100.
2. Click on "Add To My Spools"
3. You will be taken back to the Filament List.
4. The spool is not in "My Spools". Not in the list, not findable by searching "My Spools".

**Expected behavior**
I expect spools that I choose to add to "My Spools" to be there. ;-)

**Screenshots**
N/A.

**Desktop (please complete the following information):**
 - OS: Win11, latest GA
 - Browser: Chrome
 - Version: Version 131.0.6778.205 (Official Build) (64-bit)

**Additional context**
The issue happens frequently enough that I encountered it at least once during each spool-adding session that I did so far. Having it happen multiple times in a row, as now, is the first time.

markcarroll, 2024-12-23:
is it on a second page? I am trying to reproduce this and so far no luck.

luckygreen, 2024-12-23:
I just tried it again. It is still not working for me.
1. Go to the main "Filament List" page.
2. Type in "70100", because I am lazy and I hope whomever added the filament to the DB put the item number in () to the right of the color. (And if I can't find a color that way, then I edit the color name to include the item number for the next person searching).
3. I click on the filament to open up the detail page for that filament.
4. I click to add to My Spools.
5. I am taken back to the Filament List. Meaning the click clearly took.
6. When I search "My Spools", there is no filament 70100 there.

I just tried this for "15200" (Glow Pink). Same issue.

What do you see when you look inside the DB for My Spools? Do you see a 70100 and 15200 in the DB itself? Perhaps this isn't an issue of adding to the DB, perhaps this is an issue of retrieving from the DB. Such as the front end pulling from a DB cache that isn't being updated. Just a thought...

markcarroll, 2024-12-24:
ok that use case is what I needed.  found the issue.  There is still some inconsistency from NextJS that I am not happy about, but the particular problem you found was definitely my error.  Fixed now.

closed 2024-12-24 by markcarroll