MarksMakerSpace/filament-profiles #64: 💡 [FEAT] Indicate spool location on My Spools page.

mikebronner, 2024-12-24:
**Describe the solution you'd like**
Would love to have an indicator column on the My Spools page that shows an icon representing shelf, printer, or AMS, which when hovered over shows any details (name / description).

markcarroll, 2024-12-24:
I like this idea. I will play with some options to see what works well.

mikebronner, 2024-12-24:
That would allow me to quickly sort by printer/ams to group filaments by location

markcarroll, 2024-12-25:
Went to implement this and then realized that one row on my spools can represent multiple spools. For example I have three spools of Bambu white filament, one in an AMS and the other two on a shelf. What would we show in this case?

mikebronner, 2024-12-25:
Hi Mark,

Would about basing the My Spools page on the My Spools records, instead of on the filament records? That way you would always have one-to-one, and we could see remaining grams per spool. I think as you build out more options to track individual data points per spool as many other feature requests suggest, doing it this way will provide greater flexibility.

Would we loose functionality by showing each individual spool instead of showing just the filament entries? (I don't see anything lost, but there may be aspects that you intended that I am missing.)

markcarroll, 2024-12-30:
Funnily enough, Mike, I had it that way to start with and then switched it to the summary view. I may well just add both so people can choose which they want to see.

markcarroll, 2025-01-29:
There is now a column that you can turn on that shows the Location on the main My Spools page.  It is currently off by default since it makes the page a lot busier.

closed 2025-01-29 by markcarroll