MarksMakerSpace/filament-profiles #65: 💡 [FEAT] Search by color

mikebronner, 2024-12-24:
**Is your feature request related to a problem? Please describe.**
I often need to match filaments to existing colors used by brands.

**Describe the solution you'd like**
It would be amazing to be able to search by a given hex color, and return filaments that are similar in color. Sure, this would require a bit of color theory and math, but should be doable.

**Describe alternatives you've considered**
https://filamentcolors.xyz tries to do this, but its inventory is limited, and their backlog is ever growing with no end in sight.

**Additional context**
There may need to be an additional flag to indicate the filament color has been verified, meaning that it matches the actual color, and not a best guess provided by a user as a placeholder.

markcarroll, 2025-02-25:
@mikebronner I have good news for you.  I am preparing to deploy this right now.  

https://github.com/user-attachments/assets/b13d5771-acdc-4bef-8d6c-8f803c63491e

markcarroll, 2025-02-28:
closing this as I think this is what you were looking for

closed 2025-02-28 by markcarroll

mikebronner, 2025-02-28:
very cool!