MarksMakerSpace/filament-profiles #66: 💡 [FEAT] TD on Label

BadPinnochio, 2024-12-25:
**Is your feature request related to a problem? Please describe.**
I'd like to see the transmission distance on the label instead of having to look it up.

**Describe the solution you'd like**
Add the TD value to label, similar to have you have the temps or under the hex color.

**Describe alternatives you've considered**
I have edited labels to add this myself. 

**Additional context**
Example:
[image]

markcarroll, 2024-12-25:
what do you think about replacing the Pressure Advance with TD?

BadPinnochio, 2024-12-25:
I definitely find the TD to be more useful when I’m looking at the label so
it would work for me.

markcarroll, 2025-05-07:
Just released an update with selectable fields for the label. I added TD to this list.

closed 2025-05-07 by markcarroll