MarksMakerSpace/filament-profiles #74: 💡 [FEAT] Use case-insensitive sort for dropdowns

akurtz, 2025-01-06:
It appears that dropdown lists use a case-sensitive sort which breaks my expectation that brands are sorted alphabetically.  For example, "eSUN" is at the very bottom after the "Z" section instead of in the "E" section, and "Bambu Lab" is 8th in the "B" section where it should be 1st alphabetically.  This also makes it difficult to quickly jump to a brand by typing part of its name, because that browser feature is case-insensitive.

[image] [image]

Not sure if this was intentional, so I'm calling this a feature request instead of a bug.  I assume all lists use the same case-sensitive sort, but I haven't noticed it anywhere else because of the consistent Title Casing of the material list entries.

markcarroll, 2025-01-06:
you are absolutely correct in calling it out, it should not be that way.  I have a big database update coming that I am about to deploy.  Amongst the updates is a fix to the sorting issue.  This update is a big data model refactor so I am waiting for a quiet time to deploy it.  

Thanks for taking the time to log the issue.  I hope you are happy with the update.

markcarroll, 2025-01-07:
closed with today's deployment
[image]

closed 2025-01-07 by markcarroll

akurtz, 2025-01-11:
Thanks for the speedy updates! 🚀