MarksMakerSpace/filament-profiles #77: 🐛 [BUG] Filament ID 50 not loading

minsis, 2025-01-08:
**Describe the bug**
When click on https://3dfilamentprofiles.com/filaments/50/details the filament doesn't load.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://3dfilamentprofiles.com/filaments/50/details

**Expected behavior**
Filament site loads

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS M2
 - Browser safari
 - Version 17.6

markcarroll, 2025-01-08:
that's odd. It works for me, clicking on your link:
[image]
I would love to try to find the root cause if you have anything else you can give me.

minsis, 2025-01-08:
Oddly the link is now working for me now. Before it would just go to that one, and I would select a duplicate color and it would load fine. If it happens again I'll raise it up and try to get more client side logging with a HAR file or something.

markcarroll, 2025-01-08:
thank you.  I did find a bad amazon link on that page which I fixed. trying to force the issue now so I can debug it.

markcarroll, 2025-01-08:
thanks for reporting that. There was an issue with some bad actor Amazon URLs.  I have fixed the data now and will be pushing a code fix imminently too. Closing this now as the pages are fixed.

closed 2025-01-08 by markcarroll