MarksMakerSpace/filament-profiles #78: 💡 [FEAT] Improved filtering

minsis, 2025-01-08:
**Is your feature request related to a problem? Please describe.**
Not really a problem, but more of a UX issue. It would be really nice to be able to do better filtering as well as filtering backwards with breadcrumbs.

**Describe the solution you'd like**

1. There's room for improvement in adding search bars to the drop downs to help find something
2. There's room for improvement in adding drop-down over table columns in order to search something
3. Breadcrumbs, or filter tags that you can click off would be extremely helpful. As an example, if I click on PolyTerra PLA filament and then my next search needs to be PolyLite PLA+ I have to basically start my search all over again.
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

markcarroll, 2025-01-08:
Thanks for the suggestions. I agree that searching can be cumbersome right now.  If you have a suggestions of reference UIs that I can look at for ideas, please let em know.

minsis, 2025-01-08:
* I'm hardly a web developer, but in past projects at work I've used bootstrap-table which allows you to add drop-downs or pure search box to each column header that lets you filter the data.
* Other filters might be the classic shopping site filters where you have check boxes on the side that you can select/deselect to show the data you're looking for.
* Some kind of classic breadcrumb would help, where you see something like BRAND > MATERIAL > TYPE with links that let you go back to a specific selector.
* If anything a slightly better improved workflow for finding a filament could be implemented. Right now the workflow goes: BRAND > MATERIAL > TYPE, which doesn't make sense to me. As an example, PolyTerra comes in PLA and PLA+, so I'm searching material types first before I'm searching the brand type. With the brand type selected first you can simply go back and select a new material type for that type.

Those are things off the top of my head, I might have more later 😄

markcarroll, 2025-01-26:
ok so looking at this again, have you tried clicking on the names at the top? They are already a breadcrumb-like feature. 

https://github.com/user-attachments/assets/5a770e19-f076-4d76-93da-db47b86be7f6

minsis, 2025-02-02:
Ah yeah I didn't realize they were links. I still think there might be room for improved filtering on the main table without being redirected to new pages that you have to go back on. But for now the links are very helpful for going back to a specific section.

markcarroll, 2025-02-03:
I’ll see if I can make the links more obvious.

closed 2025-02-03 by markcarroll