MarksMakerSpace/filament-profiles #79: 🐛 [BUG] Data integrity issues really needs improvement

minsis, 2025-01-08:
I think the idea of having people contribute filaments and data is a great way to collect data and get the community working on a project. However, this also leads to bloat, incorrect data, and duplicates. The real big issue is the duplicates, since you have QR codes that link back to these filaments.

I think there needs to be some kind of improved validation of data in some way. Not sure what this might look like but I just wanted to at least bring this up in a ticket.

markcarroll, 2025-01-08:
Yes, I completely agree.  I have been working towards a solution here.  I want to put a unique constraint on the brand/material/type/color but cannot do so yet as there are too many duplicates.  However, as you surmised, deleting them may render people's QR codes obsolete, so I cannot do that.  At the moment deleting a filament just hides it so the QR code still works. What I am working on, though,  is a remapping table so that I can truly delete duplicates and remap them to the "master" version for that color/make combo.   

I am also changing the QR code links to point to a redirect table so that we can move what the QR code points to so it continues to work after the original filament is removed. 

Yesterday I deployed a huge data model update to the underlying database in preparation for these changes. It is going to be a multi-step process as I want everything that already exists to continue to work.  Bear with me, we will get to a better model soon.

GE-Axiom, 2025-01-11:
Just a thought, but what if we had an approval process before new filament can be added the community can put the requests in, then once someone Mark or someone he has designated approves the filament once they have ensured it isn't a duplicate and and its a legit filament? Understand we are limiting it to known brands for adding the filament but were seeing brands pop up every day. My brother ordered me some SUMLAND filament and i have never heard of it and its not in the Data base yet. SO I dont have a method to add that. If we had someone approving the filament request we would be able to allow new brands to be added and the person approving could check the info. Just a thought and im willing to help as well.

markcarroll, 2025-01-11:
Interesting idea @GE-Axiom. I do like the idea of sharing the approval burden.  As for adding brands, you can do that yourself now on the [Brands](https://3dfilamentprofiles.com/brands) page.

3dMarkK, 2025-01-13:
@markcarroll, I was coming here to report this issue too and saw the existing thread. I'd be happy to contribute some help as an approver or with the database model (I have far more years of tech/database than I do with filament, so happy to help if needed).

markcarroll, 2025-01-13:
thanks for the offer @3dMarkK do you have a Reddit account or Discord I can message you on? Hit me up on u/SirThunderCloud on Reddit

minsis, 2025-01-19:
The only downside to an approval process - even a community driven one, is gatekeeping. Unfortunately, everyone has lives and eventually ignore, or don't have time to review. We've seen this even on some of the biggest GitHub projects.

With that said I'm not 100% of a surefire way to go about it without some kind of approval process. Coming from a ServiceNow background there could be something implemented for notifications of all parties and then an auto-approval process kicks in after x number of days.

Taraan03, 2025-02-19:
Additionally, if an approval process is in place, it would be beneficial for the user who added the new filament/brand/etc. to be able to use it right away for their own spools, and the approval process would prevent "publishing" the new content for others to use. I've only just started using this tool, and I've had to add a few filaments already, and was happily surprised I didn't have to wait to have the filament in my own collection after adding it, I assumed there was an approval process since it's really hard to keep user input consistent and accurate for a tool like this.

myrrien, 2025-02-22:
This please, I added a filament color I own since it wasn't in the db and someone changed it completely so I found in my spools a color I don't own

minsis, 2025-03-02:
What determines if something gets a blue check? As for @myrrien's issue it could've been mistakes corrected. When I get a new filament on the printer first thing I do is print out my sample and go get my label, but I'll notice the filament has some incorrect data where I will correct them.

Case in point, just now I was going to get Polymaker Matte Army Blue and the two of them would be the old packaging of PollyTerra and the new Polychroma Matte. Both of these have different color codes entered, then when in fact they are the same.

What I've noticed is a lot of times the Amazon color code doesn't match what's on the manufacture's website, but since people are most likely pulling their data form Amazon (I know because I did this until I noticed the discrepancies) it ends up being wrong.

Maybe a blue check mark for the validated filament from trusted users, and then if someone edits the filament it loses its checkmark until it gets validated again.

markcarroll, 2025-03-02:
I know at least Polar Filament is putting their filament in here themselves and then referencing the polar page on their company site, so that one I know I could mark as manufacturer approved.  I think I could start working through some other brands the same way. Especially for the smaller ones that may not publish as much on other mediums. 

As far as colors go, the person that runs the filament colors site that we link up to tells me that the manufacturers he has spoken to agree that their color codes are often just a guide and not indicative of what the end result turns out to be. So he feels that the colors he measures more accurately represent the actual filament.

myrrien, 2025-03-02:
> As for [@myrrien](https://github.com/myrrien)'s issue it could've been mistakes corrected.

It wasn't a correction in my case, the spool was changed from a tricolor silk to a simple matte so completely different type and color, and this shouldn't happen. I understand adding a link or a hex code where there isn't one but drastically changing the core data isn't something that should be able to be done.

markcarroll, 2025-03-02:
@myrrien I have locked that out now.  The only thing you can change after the fact is the color. I also posted a note indicating that people have that color in their My Spools so not to fundamentally change the color, tweaking is still useful I think.

markcarroll, 2025-08-04:
This is in essence what happens.  There is a set of parameters for the "default" profile for the filament.  Each color can then override the "defaults".  I have started working directly with some manufactures who are publishing their data for me to import on a schedule into the database. I plan on adding a "manufacturer approved" type tag on those and locking them.  

There are a few fields for a user to override per spool, but I do not yet have the option to have per-user filament settings.  Nice idea though.  I will think about it some more now I have caught up on some other items.