MarksMakerSpace/filament-profiles #86: 💡 [FEAT] Brand logos collection

ch33tah0, 2025-01-18:
I prepared some logos... https://github.com/MarksMakerSpace/filament-profiles/issues/60  https://github.com/MarksMakerSpace/filament-profiles/issues/85

1. All of them are PNG (big thanks for brands that use svg or high res PNG files!)
2. Not all of them are "high resolution", but they should work. (some ebay/Amazon no name Brands do not care at all?)
3. Brands with high "filament count" got some more attention.. brands with "0" count, did not yet.

You may want to have a look at this PDF first (start from the right "1. original":
[3dprintprofiles_brand_logos_info.pdf](https://github.com/user-attachments/files/18465867/3dprintprofiles_brand_logos_info.pdf)
(Note: PDF isn't that accurate compared to the PNG files..)

**Folder "4.0 max 1000px width" should be fine for the brand logo next to the QR code.**
But as you can see some are hard to read (black on dark background)
Not sure if a drop-shadow will help? https://jsfiddle.net/6cw53dsp/

**Folders "5.x" are for the swatch label.**
I guess most files in 5.1 will work, but some need to be choosen from 5.2 or other 5.x folders
Example: Hatchbox, Sainsmart, 123-3D, Polar Filament, TCPOLY, ...
Dithering seems to work for some better than other (123-3D & Das Filament)?

My main reason for the whole 5.x folders is my laber printer and the software it comes with (no dither/grayscale):
[image]
(preview of printout. zoom in. some "light" colors will be cut away.)
I think the “feed the Laber printer only with black pixels” approach will also work well for others.

**Let me know, if there is something that needs some rework. (most stuff can be redone really quick with some PS actions and automation.)**

**Filenames:**
- no capital letters
- removed every Non-alphanumeric character
- replaced every space " " with a dash "-"

[1. original.zip](https://github.com/user-attachments/files/18465965/1.original.zip)
[2. PS no transparent pixel job.zip](https://github.com/user-attachments/files/18465966/2.PS.no.transparent.pixel.job.zip)
[3. arranged for better label.zip](https://github.com/user-attachments/files/18465967/3.arranged.for.better.label.zip)
[4.0 max 1000px width.zip](https://github.com/user-attachments/files/18465968/4.0.max.1000px.width.zip)
[5.1 fill black.zip](https://github.com/user-attachments/files/18465969/5.1.fill.black.zip)
[5.3 greyscale.zip](https://github.com/user-attachments/files/18465970/5.3.greyscale.zip)
[5.4 border in 5px then greyscale.zip](https://github.com/user-attachments/files/18465972/5.4.border.in.5px.then.greyscale.zip)
[5.6 dither 1000px transparent.zip](https://github.com/user-attachments/files/18465973/5.6.dither.1000px.transparent.zip)

A quick overview:
[image]

Searcherd, but no (good) logo found:
Brand |
-- |
Yuaneang
Likesilk
CMACAIV
HP 3DF
HUTI
PEAKACE
H3D
3D Warhorse
Amoybaby
Carbon
Color Me   3D
COLORFUL
Made for   Prusa
Menetee
PINRUI
PLA max
ProRock
TTYT3D
Yoyi   Yoyi
Cerprise
Kai   Parthy
Unicofil

Not searched yet:
Brand | count
-- | --
3D Best-Q | 0
3D Hero | 0
3D   Passion | 0
3D Printlife | 0
3D   Solutech | 0
3D Universe | 0
3DF   Filaments | 0
3DTomorrow | 0
3DXTECH | 0
AIO Robotics | 0
Alien3D | 0
APrintaPro | 0
Atomic   Filament | 0
BAMtack! | 0
bioFila | 0
Breathe-3DP | 0
California   Filament | 0
Capricorn | 0
Carbodeon | 0
CAZ3D | 0
Cocoa   Press | 0
Colorthread | 0
Colour   Dream | 0
CooBeen | 0
Copier2Go | 0
CraftyBoy | 0
Denif | 0
DR3D | 0
Dremel | 0
DUDV2 | 0
E3D | 0
Enotepad | 0
Eolas   Prints | 0
EUMAKERS | 0
FF   SUNSENKJ | 0
FiberForce | 0
Fil X | 0
Filacrafts | 0
FilaMatrix | 0
FilamentOne | 0
Filamentworld | 0
Filastruder Veracity | 0
FoxSmart | 0
Francofil | 0
Fusion   Filaments | 0
GL Robotics | 0
Goedis   3D | 0
GOO | 0
Graft   Milk | 0
Halo 3D Filaments | 0
Hello3D | 0
Hobby King | 0
HZST 3D | 0
IC3D | 0
ICE | 0
IkSystem | 0
Innofil   3D | 0
LANDU | 0
Maker   Geeks | 0
MakerGear | 0
MakeShaper | 0
MeltInk3D | 0
NatureWorks | 0
NHH | 0
Nobufil | 0
NVSMART | 0
Patona | 0
Plas3D | 0
Polyalchemy | 0
Printbed | 0
ProtoParadigm | 0
Prusa | 0
Purement | 0
REC | 0
Reflow   Filament | 0
Repkord | 0
Rhino 3D | 0
Sakata3D | 0
Sandoz   3D Filaments | 0
SeeMeCNC | 0
SEGADEN | 0
Shengtian | 0
SnoLabs | 0
SunTop 3d | 0
Tactink | 0
TECSONAR | 0
Tianse-3D | 0
Toner Plastics | 0
Treed | 0
UltiMachine | 0
U.S.   Monofilaments | 0
VO3D | 0
Voxelab | 0
WYZ works | 0
YOYI | 0
Zhuopu | 0

markcarroll, 2025-01-24:
this is fantastic, thank you for working all this out.  I am wrapping up some other work right now but I will attempt to get to adding these this weekend.

bjlled, 2025-01-25:
Omg this is amazing.  !!

markcarroll, 2025-01-29:
I've added the first few of these and they look good. Thanks again for this effort.

ch33tah0, 2025-01-30:
@markcarroll any format you prefer?
I guesss 4.0 and 5.1?

[image]
(since this is all black, it will be the same for 4-5.4)

markcarroll, 2025-01-31:
@ch33tah0 I can't thank you enough for helping with this.  As you have hopefully seen, I have been trying to add a few in the evenings when I can.  So far I have been using the 4.0 and 5.1 as you already ascertained. I hav been trying to do them one by one for now as I am trying to check what works best.  

I had realized this week while driving somewhere (and hence forgot to message) that maybe 1000px width is too big and we should standardize on 240px wide for the web version since that is the max-width for that one, the printer version doesnt really need to go over 400px since most label printers max out at 203DPI which equates to 319 pixels for a full width label. 

Your thoughts?

ch33tah0, 2025-01-31:
I absolutely understand. Downsizing images is absolutely no problem for me. This also reduces traffic minimally as a nice side effect. 😄

I can resize the 4.0 version to 240px width and add "-web" -> inland-web.png -> will become 4.1 (240px)
I will go through the 5.1 / 5.3 / 5.4 versions and select a "best of" version -> will become 6.0 (1000px) -> will become 6.1 (400px)

In the future, I will upload 4.1 and 6.1 files/zip only.

Any ideas on how to solve this (in a good and easy way) for new brands?
If everyone could upload a logo for the created brand, this would certainly not end well... (NSFW? Not standardized, etc...)

What would be the simplest workflow at the moment?
You get the two correctly named files here. -> you drag and drop into ‘logos’ folder via ftp? end -> if a logo.png is there, it will be displayed. If not, text will be generated?
You don't do every brand logo by hand, right? (I have seen the bambu-lab.svg)

Todo this weekend:
There are some 1:1 ratio brands right now (see overview first post)
SPOOL3D (I will just cut away the circle and keep -> SPOOL3D CANADA)
RS PRO ([will change to this](https://res.cloudinary.com/rspoc/image/upload/f_auto/q_auto/v1667268520/RS_PRO_CMYK_RED_horizontal_primary.jpg))
bq (nothing i can do, bq people have to live with this small logo 🤔)
Kodak? ([I think I will change to this](https://commons.wikimedia.org/wiki/File:Logo_of_the_Eastman_Kodak_Company_(2006%E2%80%932016).svg))

5.1 not optimal / other rework, mostly bc white pixel mix with other colors:
Hatchbox
3dkordo
sainsmart
123-3D
das filament
polar filament
american filament
bing3d
polaroid
TC Poly
...

markcarroll, 2025-01-31:
Sounds good on the sizing and thank you, I will wait for 4.1 and 6.0 versions before I do any more. I am not convinced we need to go as large as 1000px for the printed ones. 800px would be the maximum I think is needed.  

I think for now a GitHub request is fine for new brands. We doin't get a lot of new ones since we have 345 now, how many more can there be? 😄 

At the moment it is indeed manual. I have been opening and resizing, if needed, each one individually.  Then I put the file names in a JSON file that is used to load them at runtime.  Obviously this is tedious and doesn't scale, so having consistent file names and sizing will help dramatically. I can then just write a script that takes the file names and puts them in the JSON file. 

Funnily enough I did actually write code and DB table to accept uploads for brand logos, but then backed off of that idea precisely because of the issues you have already mentioned.  I think that what we are discussing here is a much safer approach. 

BTW, the back logos are not ideal on the dark model background but I thought the outline seemed to do a little dis-service to their branding as it did change it fairly dramatically.  I did winder if there was a way to put a blurred inverted color version behind it that helped it pop, but the results were not consistent, so I stopped. 

[image]

[image]

ch33tah0, 2025-02-04:
[4.1 -web.png (240px).zip](https://github.com/user-attachments/files/18662541/4.1.-web.png.240px.zip)
[6.1 best of 5.x (400px).zip](https://github.com/user-attachments/files/18662543/6.1.best.of.5.x.400px.zip)

added like ~35 more brands.

4.1 -web.png (240px).zip -> added the "-web" to name and resized to max 240px width:
[image]

---

6.1 best of 5.x (400px).zip -> made some adjustments for label printer:
[image]

---
btw:
https://3dfilamentprofiles.com/filaments/recyclingfabrik
https://3dfilamentprofiles.com/filaments/recycling-fabrik
recycling(-)fabrik.png and bambu-lab(s).png should be the only two duplicate files. -> so technically 229/351 brands done 👍

---
ProRock only image of a brand logo I could find (BUT 1000px, good pictures and good lighting!):
[image]
so I made this.. close enough? 😂
[image]
I found some high res logos in PDF data sheets... but not on the homepage or on the WWW... crazy

---
Never heard of Tailwind, but seems to be CSS without using CSS... CSS easy mode? 😄
The secret for consitent blur is to use only 1-2 px?
https://jsfiddle.net/0asnxgz8/2/ (change the 30px to 1 or 2px)
Thin lines like in the polymaker logo do not work well with a high blur value.

---
let me know if you need any other resolution.

markcarroll, 2025-02-05:
That came out pretty well for an angled picture.  

Ok so I have been playing around. There are times where the web label _could_ stretch up to 480px x 64px so if we wanted to provide images at that resolution I don't think it would be too onerous. The label folder is only 2.5MB now so the web one would be just a little bigger but not bad really. Of course we will have to see what that does to my data egress bill. 🤣  
I'm quite happy to just use what I have if that is easier too, I hate to ask for more. 

Also, what do you think about this:

No drop shadow:
[image]

With your 30px drop-shadow code:

[image]

[image]

[image]

[image]

This is with 1px drop shadow:

[image]

This is the background color we are using in dark mode: `#1e293b`

I deployed to production so let me know what you think

ch33tah0, 2025-02-05:
I will leave the 480px version here, use it if you want. 
But I think the 240px version is just fine! (480px increase from 2.38MB -> 4.27MB)
[4.2 -web.png (480px).zip](https://github.com/user-attachments/files/18668375/4.2.-web.png.480px.zip)

I think the current 1px 0.49 shadow is perfect!

But you may want to add a black shadow for **light mode** (I'm blind now 😎)
https://3dfilamentprofiles.com/filaments/2570/details
[image]

markcarroll, 2025-02-06:
https://www.reddit.com/r/3dfilamentprofiles/s/CWCC4o93bT

Amoenus, 2025-03-06:
I grabbed a couple from the "unsearched" list for you to process
Plas3D

[image]

Patona

[image]

DR3D

[image]

Capricorn

[image]

Inofil 3D

[image]

Unsure this is the best way to submit those though

exiom-xyz, 2025-05-12:
[image]

Hi! I have recently added Alchemakes into the database, here is the logo for your reference... much thanks!

markcarroll, 2025-05-13:
@ch33tah0 any chance you could do your magic to these more recent logos?

exiom-xyz, 2025-05-14:
@ch33tah0 I am guessing that it is done this way because you want to keep a certain standard/consistency for graphics... 
perhaps a guideline should be published so most of the work is done and others can submit a ticket or even a pull request instead?

markcarroll, 2025-05-14:
@exiom-xyz great idea on PRs..   @ch33tah0 I have added the logo folders in this repo in `/logos`. There is one folder for `labels` and one for `web` which are used as the name implies. The `label` folder logos are for the swatch labels, designed to be printed on 203DPI black and white label printers, and the `web` folder logos are for the Brand card on the web pages. There is also a `logo.json` file with each brand name that has a logo listed. This is used in the code to know what files to load, so will need to be updated to reflect any new brand logos that are added.  LMK if you need more explanation. 

Now folks can just submit a PR with new logos.    @ch33tah0 if you can put together a quick spec I will add it to the README. 

Thanks again for all your help.

exiom-xyz, 2025-05-14:
Submitted Alchemakes PR as a test, let me know if it needs adjusting and/or conform with the rest of the logos
https://github.com/MarksMakerSpace/filament-profiles/pull/133

markcarroll, 2025-05-14:
@ch33tah0 that guide is awesome, thank you. I will upload as soon as I can - it is my birthday today so i might not get a chance until tomorrow. 

@exiom-xyz thank you for the PR.  The test will be to see how it prints out on a label.  I suspect the label version is still too complex for it to come out cleanly with the low resolution label printers that most people have.  I am happy to give it a go when I get a chance and report back.

markcarroll, 2025-05-16:
@exiom-xyz I merged your PR but it looks like it didn't meet @ch33tah0 's spec.  Specifically the background is not transparent, and I do believe the small characters under the main logo may not work on a label printer. 

On the plus side, there is now an automation in place to handle the logos once the PR is merged.

Robert-B-H, 2026-05-05:
@markcarroll do you wish to keep this thread open for Brand Logo standardization discussions and additional mass efforts in populating logos or do you prefer to close this and move any discussions to Discord?