MarksMakerSpace/filament-profiles #88: 🐛 [BUG] Text being truncated with long filament types

minsis, 2025-01-19:
**Describe the bug**
Text on the label is truncated for longer filament types.

https://3dfilamentprofiles.com/filaments/3340/details

minsis, 2025-01-19:
This seems to be happening for label life default image and PNG, but not the native format.

Default:

[image]

Native:

[image]

PNG:

[image]

markcarroll, 2025-01-21:
this is now fixed

closed 2025-01-21 by markcarroll

minsis, 2025-01-25:
This is still an issue: https://3dfilamentprofiles.com/filaments/3629/details

markcarroll, 2025-01-25:
ok @minsis I have updated the function to resize that box yet again (can they get any longer?).  Now it maxes out at 30 characters and after that truncates the string, since I think we get to the point it becomes illegible if we go any smaller.  Refresh the page and let me know if that is looking any better.

minsis, 2025-01-26:
These day who knows 🤷. Polymaker is rebranding their entire lineup with Panachroma. Just doing a website spot check it looks like maybe the longest name they have is `Panachroma Gradient Matte`

markcarroll, 2025-01-26:
I think we might also need to truncate the PLA+/PRO to just PLA+ for longer labels. Sorry for the delay on getting this back out, but I screwed up the database on that last deploy that you caught. Think it is fixed now. Serves me right for relying on AI to help redo a DB query while running a fever. Moral - never trust the AI.

minsis, 2025-01-26:
It's all good. I also agree about maybe changing it to PLA+.

Not sure what point where you cut off the readability of the label and the need for having all the numbers. Not sure if there's any kind of way to abbreviate some of  more lengthy words for the label. Similar applications do things like this where they have a word, the pluralized word, and an abbreviated word all associated to said word.