MarksMakerSpace/filament-profiles #89: 💡 [FEAT] Limit website links to manufacture only URL's

minsis, 2025-01-19:
**Is your feature request related to a problem? Please describe.**
Coming from issue #77 where you stated an erroneous amazon link that was causing a site issue. I've noticed that a lot of filaments "websites" are being linked to Amazon and not the actual manufacture's website.

**Describe the solution you'd like**
To improve security it might not be a bad idea to lock the website field hostname to only match the manufacture's website. Since you're already asking for Amazon AISN number, the need to link Amazon in the website is a moot point.

**Describe alternatives you've considered**
The only other alternative is for users to constantly have to catch these issues and be willing to change them.

markcarroll, 2025-01-19:
There definitely needs to be some clean up of the website link URLs, however there are almost 3,000 records now, so that is going to be a lot of work.  I have thought about removing Amazon links altogether, but then that field will be empty for a large number of filaments.  Maybe that is ok, but I am erring on the side of having _something_ in that field rather than nothing.  

I am happy for someone/anyone to correct any issues that are found.  

I may put a warning up in the new filament form that prompts you to put something else though. That seems easily doable.

minsis, 2025-01-19:
Better an empty field than a potential erroneous amazon link.

There can still be an enforcement so that when someone does edit one they're forced to remove the amazon link and swap it out for something else.

The other option might be to just remove all amazon related links and replace them with the base manufacture website - not necessarily the filament.

markcarroll, 2025-01-20:
If there is no link in the filament it will use the one from the “default” for that brand/material/type, so maybe we erase all the Amazon links and just work on populating the defaults.  That would cover things pretty quickly.

minsis, 2025-02-02:
> There are a few cases though, where the Amazon link is there because the filament is for a different country and right now the Amazon price only searches and goes to the US site. I feel like that is something I need to resolve first.

Yeah this is partly what I'm talking about as an associated "approved" domains for a particular maker. i.e. if it's an amazon brand then that URL field allows for amazon.com only URL's.

markcarroll, 2025-02-06:
The code now automatically removes bad Amazon links and I have also gone through and updated a lot of these to point to the manufacturer page.  There are still quite a few mere to do but those will happen over time.  Closing this ticket for now as the main concern should have been mitigated.

closed 2025-02-06 by markcarroll