MarksMakerSpace/filament-profiles #92: 🐛 [BUG]

minsis, 2025-01-25:
**Describe the bug**
Some Labellife native downloads are crashing Labelife. I've added a new filament where the name is being truncated in the image version, but now the native format is crashing Labelife. https://3dfilamentprofiles.com/filaments/3629/details

Looking at the crash report I see the following:

```
"exceptionReason" : {"arguments":["-[__NSArrayM insertObject:atIndex:]"],"format_string":"*** %s: object cannot be nil","name":"NSInvalidArgumentException","type":"objc-exception","composed_message":"*** -[__NSArrayM insertObject:atIndex:]: object cannot be nil","class":"NSException"}
```

Not sure if this is meaningful.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to this newly created filament profile: https://3dfilamentprofiles.com/filaments/3629/details
2. Change to native format and click download

Try some other filaments where names are not truncated and they're fine.

**Expected behavior**
Downloaded native labelife format should not crash the app.

**Screenshots**

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser safari
 - Version latest

markcarroll, 2025-01-26:
turns out it doesnt like empty strings. Changed the empty fields to spaces and it seems to work fine.  This will be fixed in the next deployment.

closed 2025-01-26 by markcarroll

minsis, 2025-02-02:
@markcarroll 👍  Seems to be working now. Thanks!