MarksMakerSpace/filament-profiles #93: 🐛 [BUG] My Spool links are going to a 404

minsis, 2025-01-25:
**Describe the bug**
Clicking on a link from the My Spools page is generating a 404.

**To Reproduce**
Steps to reproduce the behavior:
1. My Spools
2. Click on any spools saved
3. Witness a 404

**Expected behavior**
I should be going to the filament page of my saved spool.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: macOS
 - Browser Safari
 - Version latest

markcarroll, 2025-01-25:
well crap, that last deployment didnt go right. I have rolled back for now so your other label issue will not be fixed until I solve this one. thanks for catching.

markcarroll, 2025-01-26:
ok this and your other label issue should be fixed now. Phew! 😰

closed 2025-01-26 by markcarroll

minsis, 2025-02-02:
@markcarroll 👍 Working nicely now! Thanks!