MarksMakerSpace/filament-profiles #95: 💡 Option to provide feedback/correction to filament entries with incorrect or missing information.

christopherstavrou, 2025-01-27:
**Is your feature request related to a problem? Please describe.**
There could be many filament entries with missing information for example, BambuLab Neon City (Blue-Magenta). It is missing all information and the color.

**Describe the solution you'd like**
Allow an option from the filament page to request a change that updates for all users the default information that is provided for the filament.

**Describe alternatives you've considered**
This could be done via the create new issue in GIT but the risk is that this method could bloat the ticket system and it seems inefficient.

markcarroll, 2025-01-29:
On the details page you can click on the brand, material or type and add default settings for all filaments of that type. 

[image]

markcarroll, 2025-01-29:
If your request is for someone else to find the data, I think maybe we could do that in the Discussions section of this site

christopherstavrou, 2025-01-30:
> On the details page you can click on the brand, material or type and add default settings for all filaments of that type.
> 
> <img alt="Image" width="628" src="https://private-user-images.githubusercontent.com/599566/407602489-04cbaf2d-cb50-48de-9594-a27f48f8b8f6.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzgyNDU0NTgsIm5iZiI6MTczODI0NTE1OCwicGF0aCI6Ii81OTk1NjYvNDA3NjAyNDg5LTA0Y2JhZjJkLWNiNTAtNDhkZS05NTk0LWEyN2Y0OGY4YjhmNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTMwJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDEzMFQxMzUyMzhaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zMTU2MDZhY2E0MTI3YTcxOTg4ODRkYjUxODI2YmNkOTE4Yzc2NTdmNTI4OGU4MjYwMTFhNGU0OTk2MmYzYjg0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.V1Fw4mA7WayOpZtTba57AGaqirEsQtwEzzSFJKEo5KA">
> https://3dfilamentprofiles.com/defaults/bambu-lab/pla

This is good to know and satisfies my use case mostly! Thanks for the response.
The feature request is to update the default data for all users if no data exists as is the case for this filament entry. It might not be necessary given your solution so I will close this ticket.

closed 2025-01-30 by christopherstavrou