MarksMakerSpace/filament-profiles #97: 💡 [FEAT] Add printing tracking system

pretorien51, 2025-01-29:
**Is your feature request related to a problem? Please describe.**
Interface where to add prints and automatically decrease spool quantity

**Describe the solution you'd like**
Add a section where we can add object that we printed and indicate how many filament we use to decrease automatically spool filament quantity. This could allow us to estimate easily filament status on each spool.

**Describe alternatives you've considered**
I saw this feature on Spoolstock, clearly inspired by this site. But that tracks all my prints and can even estimated cost (each spool have a price)

markcarroll, 2025-02-08:
working on it.

markcarroll, 2025-02-12:
just deployed this: https://3dfilamentprofiles.com/my/print

I will continue to grow it but hopefully that covers your request.

closed 2025-02-12 by markcarroll