MarksMakerSpace/filament-profiles #100: 🐛 [BUG] Not able to connect with My Printers

sxdjt, 2025-02-07:
**Describe the bug**
When trying to add my printer to "My Printers", the system generates an error, "Unable to fetch printer data.  Did you add".

**To Reproduce**
Steps to reproduce the behavior:
1. Go to https://3dfilamentprofiles.com/my/printers
2. Fill in the local HA URL and LLAT
3. Click Save
4. Observe the error message

I'm assuming the "Did you add" meant to ask if the CORS section had been added to configuration.yaml, which it has.  HA was restarted after adding that section.

**Expected behavior**
It works?

**Screenshots**

Error message:

[image]

**Desktop (please complete the following information):**
 - OS: macOS Sequoia 15.3
 - Browser Firefox 134.0.2 and Edge 128.0.2739.63
 - Version [e.g. 22]

markcarroll, 2025-02-07:
Yes sorry about the error message.  Have you tried Chrome? That is what I have been using.

sxdjt, 2025-02-07:
The only chromium browser I have is Edge.  I can download Chrome and give it a try though.

markcarroll, 2025-02-07:
It’s an SSL thing going from secure to insecure site.  Browsers do their damndest to prevent that now.   I’m looking for another solution.

closed 2025-02-07 by markcarroll