MarksMakerSpace/filament-profiles #99: 💡 [FEAT] Filament ID on labels

minsis, 2025-02-07:
It might be nice to see filament ID's on the labels as well, or maybe even displayed someone on the site other than URL's.

While I still print the swatches, I  basically like collecting Benchy prints as my samples so I have a mini armada. If there was some ID of the filament then I could just sharpie that on the bottom of the print.

With that, there could be issues with it considering the duplicated filament colors and creating a redirect table for it.

markcarroll, 2025-02-08:
At some point I want to make the labels more customizable, so this will go in that bucket.

trinkets02, 2025-02-14:
Agree, having an inventory number for each filament in the database would be ideal that then could be printed to the label.

markcarroll, 2025-05-07:
Just released an update with selectable fields for the label. I added filament ID to this list.

closed 2025-05-07 by markcarroll